isaac-sim/IsaacLab #3475: [Question] Simulation App hangs when closing as does simulation context.

bkolligs, 2025-09-16:
### Question

I'm trying to gracefully shutdown my application but the two methods that I would expect to work here hang: `SimulationContext.stop()` and `SimulationApp.stop`. 

#### `SimulationApp.stop`

With verbosity increased, I see the following output after a call to `simulation_app.stop`: 
```bash
2025-09-16T19:33:53Z [70,301ms] [Verbose] [omni.datastore] Disk size in budget: 5853240, budget: 102400 MiB
2025-09-16T19:34:23Z [100,306ms] [Verbose] [omni.datastore] Disk size in budget: 5853240, budget: 102400 MiB
2025-09-16T19:34:53Z [130,308ms] [Verbose] [omni.datastore] Disk size in budget: 5853240, budget: 102400 MiB
2025-09-16T19:35:23Z [160,311ms] [Verbose] [omni.datastore] Disk size in budget: 5853240, budget: 102400 MiB
```
Any idea why this would be occurring? Thanks!

PeterL-NV, 2025-09-16:
Thanks for reporting this. Seems the issue is with `sim.reset()`. If you comment that out, `simulation_app.close()` will not hang. Will follow up.

bkolligs, 2025-09-17:
Interesting, during my application though I am calling `sim.reset()` frequently as I cycle through random positions, so how can i call it while maintaining the ability to close the app?

PeterL-NV, 2025-09-17:
We're working on a fix. I think you don't need to make any change in your application. Just your app will not close gracefully for now, you can use `Ctr + C` to close it as a WAR.

exaFLOPs26, 2026-01-12:
I think issue fixed? My code is not being terminated properly.

AntoineRichard, 2026-08-27:
Thanks @bkolligs for the minimal repro — it made this easy to re-test properly.

We ran your exact MRE shape on current `develop` (Isaac Lab 3.x / Isaac Sim 6.x), headless: `AppLauncher` → `SimulationContext` → `InteractiveScene` (dome light only) → `sim.reset()` + `scene.reset()` → `simulation_app.close()`. The process exits cleanly with code 0 in about 4 seconds, with a no-reset control run behaving identically — no hang, no SIGKILL needed.

@exaFLOPs26 — if your application still doesn't terminate on a current release, that's likely a different exit path; a fresh issue with your script would help us chase it separately.

Closing as resolved on 3.0 — please reopen if the hang returns on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard