isaac-sim/IsaacLab #3477: [Bug Report] ： Warp CUDA error: 'cuDeviceGetUuid' / CUDA error 36 with NVIDIA Driver 580.65.06 (Blackwell GPU)

wllevy, 2025-09-17:
### Describe the bug

I am encountering a Warp CUDA error when attempting to run reinforcement learning within the latest container. The error messages indicate a compatibility issue related to the cuDeviceGetUuid function and CUDA error 36, suggesting that an API call is not supported by the installed NVIDIA driver. This occurs despite using a very recent NVIDIA driver ( version 580.65.06).

### Steps to reproduce

Execute the following command inside the container to run a reinforcement learning task:

```
./isaaclab.sh -p scripts/reinforcement_learning/skrl/train.py --task Isaac-Reach-UR10-v0 --headless
```

### Actual Behavior
```
root@romiserver:/workspace/isaaclab# ./isaaclab.sh -p scripts/reinforcement_learning/skrl/train.py --task Isaac-Reach-UR10-v0 --headless
[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh                                                                                                           
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /workspace/isaaclab/apps/isaaclab.python.headless.kit
Loading user config located at: '/isaac-sim/kit/data/Kit/Isaac-Sim/5.0/user.config.json'
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/5.0/kit_20250917_040652.log
2025-09-17T04:06:52Z [2ms] [Warning] [omni.ext.plugin] [ext: msr] Extensions config 'extension.toml' doesn't exist '/workspace/isaaclab/source/msr' or '/workspace/isaaclab/source/msr/config'
2025-09-17T04:06:52Z [2ms] [Warning] [omni.ext.plugin] [ext: msr-surgical] Extensions config 'extension.toml' doesn't exist '/workspace/isaaclab/source/msr-surgical' or '/workspace/isaaclab/source/msr-surgical/config'
2025-09-17T04:06:53Z [440ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-09-17T04:06:53Z [954ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.
Warp CUDA error: Failed to get driver entry point 'cuDeviceGetUuid' (CUDA error 1)
Warp CUDA error: Function cuDeviceGetUuid_f: a suitable driver entry point was not found
Warp CUDA error 36: API call is not supported in the installed CUDA driver (in function cuda_init, /builds/omniverse/warp/warp/native/warp.cu:277)
|---------------------------------------------------------------------------------------------|
| Driver Version: 580.65.06     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA RTX PRO 6000 Blackwell .. | Yes: 0 |     | 98133   MB | 10de      | 0          |
|     |                                  |        |     |            | 2bb1      | c714f810.. |
|     |                                  |        |     |            | e1        |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-79-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101011 (1.21.1.11)
| Processor: AMD Ryzen Threadripper PRO 7965WX 24-Cores
| Cores: 24 | Logical Cores: 48
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 257197 | Free Memory: 243636
| Total Page/Swap (MB): 8191 | Free Page/Swap: 0
|---------------------------------------------------------------------------------------------|

```

-->

### System Info

- Commit: N/A
- Isaac Sim Version:  5.0
- OS: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
    Kernel: 6.8.0-79-generic
- GPU: NVIDIA RTX PRO 6000 Blackwell
- CUDA: 13.0
- GPU Driver: 580.65.06

### Checklist

- [x] I have checked that there is no similar issue in the repo 

### Acceptance Criteria

- [ ] Isaac Lab runs successfully within the  container with NVIDIA Driver 580.65.06 on a Blackwell GPU.

nathantsoi, 2025-09-18:
+1 on a 5070ti with 580.82.09

```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 580.82.09              Driver Version: 580.82.09      CUDA Version: 13.0     |
+-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 5070 Ti     Off |   00000000:01:00.0  On |                  N/A |
| 30%   44C    P1            106W /  300W |    7371MiB /  16303MiB |     69%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
```

jayant-mil, 2025-09-21:
Same here

```log
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/jayant/Repositories/IsaacLab/apps/isaaclab.python.headless.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/jayant/miniconda3/envs/isaaclab/lib/python3.11/site-packages/omni/data/Kit/Isaac-Sim/5.0/user.config.json'
[Info] [carb] Logging to file: /home/jayant/miniconda3/envs/isaaclab/lib/python3.11/site-packages/omni/logs/Kit/Isaac-Sim/5.0/kit_20250922_004448.log
2025-09-21T22:44:49Z [129ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-09-21T22:44:49Z [593ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.
Warp CUDA error: Failed to get driver entry point 'cuDeviceGetUuid' (CUDA error 1)
Warp CUDA error: Function cuDeviceGetUuid_f: a suitable driver entry point was not found
Warp CUDA error 36: API call is not supported in the installed CUDA driver (in function cuda_init, /builds/omniverse/warp/warp/native/warp.cu:277)

|---------------------------------------------------------------------------------------------|
| Driver Version: 580.65.06     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 5070 Ti       | Yes: 0 |     | 16303   MB | 10de      | 0          |
|     |                                  |        |     |            | 2c05      | d9e59032.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 24.04.3 LTS (Noble Numbat) ubuntu, Version: 24.04.3, Kernel: 6.14.0-29-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101011 (1.21.1.11)
| Processor: AMD Ryzen 5 8400F 6-Core Processor
| Cores: 6 | Logical Cores: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 63904 | Free Memory: 59509
| Total Page/Swap (MB): 8191 | Free Page/Swap: 8191
|---------------------------------------------------------------------------------------------|
2025-09-21T22:44:54Z [5,494ms] [Warning] [gpu.foundation.plugin] PCIe link width current (8) and maximum (16) for device 0 don't match.
2025-09-21T22:44:54Z [5,494ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
[INFO]: Parsing configuration from: <class 'isaaclab_tasks.direct.factory.factory_env_cfg.FactoryTaskGearMeshCfg'>
Traceback (most recent call last):
  File "/home/jayant/Repositories/IsaacLab/./scripts/reinforcement_learning/skrl/train.py", line 227, in <module>
    main()
  File "/home/jayant/Repositories/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 81, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name.split(":")[-1], agent_cfg_entry_point)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jayant/Repositories/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 45, in register_task_to_hydra
    agent_cfg = load_cfg_from_registry(task_name, agent_cfg_entry_point)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jayant/Repositories/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py", line 77, in load_cfg_from_registry
    raise ValueError(
ValueError: Could not find configuration for the environment: 'Isaac-Factory-GearMesh-Direct-v0'.
Please check that the gym registry has the entry point: 'skrl_cfg_entry_point'.
Existing RL library (and algorithms) config entry points: 
  |-- rl_games: PPO
2025-09-21T22:44:55Z [6,260ms] [Warning] [omni.physx.plugin] USD stage detach not called, holding a loose ptr to a stage!
2025-09-21T22:44:55Z [6,433ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

songgaogy, 2025-10-15:
same problem