isaac-sim/IsaacLab #3554: [Bug Report] Massive performance drop rebased to IsaacLab v2.2.1 for IsaacSim 5.0.0

mheyrman, 2025-09-25:
### Issue

After upgrading to IsaacSim 5.0.0 to accommodate a 50-series GPU and rebasing my code to the latest release, performance massively dropped. I made no changes to my env_cfg or rl_cfg aside from adding the obs_groups dictionary element to be in line with the current rsl_rl and IsaacLab releases.

There is a performance drop associated with using the new IsaacSim 5.0.0 and one associated with the IsaacLab rebase.

Simulation (headless) startup time differences:
- 5.0.0 w/ rebase:
  - Time taken for scene creation : 15.875053 seconds
  - Time taken for simulation start: 26.286755 seconds
- 4.5.0 w/ rebase:
  - Time taken for scene creation : 4.985811 seconds
  - Time taken for simulation start : 13.310909 seconds

IsaacLab collection time differences:
- 4.5.0 w/o rebase: ~ 1.2 seconds

[image]

- 4.5.0 w/ rebase: ~1.8 seconds
- 5.0.0 w/ rebase: ~2.0 seconds

[image]

Learning time has not noticeably changed. 

### System Info

- Isaac Sim Version: 5.0.0
- OS: Ubuntu 24.04
- Starting time system:
  - GPU: NVIDIA RTX 5080
  - CUDA: 12.9
  - GPU Driver: 575.64.03
- Collection time system:
  - GPU: NVIDIA RTX 4090

RandomOakForest, 2025-09-26:
Thank you for posting this. The team will review this report, and provide guidance.

mheyrman, 2025-10-07:
Mostly solved by #3636, closing issue.