isaac-sim/IsaacLab #3571: [Question] Franka Panda’s gripper do not fit tightly to its visual geometry

JyiDeng, 2025-09-26:
# Question

I am encountering an issue where the collision colliders (visible as green lines) on the Franka Emika Panda robot’s gripper do not fit tightly to the gripper’s visual geometry. This misalignment may lead to problems such as the gripper penetrating objects or incorrect collision responses.

colliders not tightly fit the gripper:

[image]

gripper penetrating objects: (the greyish part of the gripper is invisible as it penetrates the cube)

[image]

# Environments

Isaac Sim Version: 5.0.0
Operating System: Ubuntu 22.04
GPU Model: NVIDIA GeForce RTX 5090
Driver Version: 570.86.10
nvcc Version: Build cuda_12.8.r12.8/compiler.35404655_0

# Steps to Reproduce

1. Launch Isaac Sim 5.0.0 and create a new empty workspace.
2. Load Franka: Create → Robots →  Franka Emika Panda Arm.
3. Open colliders through:

[image]

4. Load cubes:  Create → Mesh → Cube, and Add → Physics → Colliders Preset, and then Add → Physics → add RigidBody:

[image]

Then, save the cube as usd, and load the usd afterwards using [RigidObjectCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.assets.RigidObjectCfg) , where I doubt **contact_offset=0.05** is somehow wrong:

```python
obj_cfg = RigidObjectCfg(
    prim_path=prim_path,
    init_state=RigidObjectCfg.InitialStateCfg(pos=[0.0, 0.0, 0.0], rot=[1.0, 0.0, 0.0, 0.0]),
    spawn=UsdFileCfg(
        activate_contact_sensors=False,
        usd_path=usd_path,
        scale=(1.0, 1.0, 1.0),
        rigid_props=RigidBodyPropertiesCfg(
            solver_position_iteration_count=16,
            solver_velocity_iteration_count=1,
            max_angular_velocity=1000.0,
            max_linear_velocity=1000.0,
            max_depenetration_velocity=5.0,
            disable_gravity=False,
            kinematic_enabled=False,
        ),
        mass_props=sim_utils.MassPropertiesCfg(mass=10.0),
        collision_props=sim_utils.CollisionPropertiesCfg(
            contact_offset=0.05
        ),
    ),
)
```

5. Let franka grip the cube through code (not so related to franka’s colliders).

# What I’ve Tried

I’ve tried to follow the steps: Go to Property panel → Collider → Approximation →Switch to Convex Hull, which is suggested by: [How to edit collsion properties in ISAAC SIM?](https://forums.developer.nvidia.com/t/how-to-edit-collsion-properties-in-isaac-sim/282806)  However nothing is changed given any options (Convex Hull, SDF Mesh, Convex Decomposition, etc.)

PeterL-NV, 2025-09-29:
I didn't see the same issue. 
Can you zoom in and check the greyish part of the gripper is actually penetrates the cube? Or you can put a larger cube along size the arm and see if it can penetrate into the cube.

JyiDeng, 2025-09-30:
> I didn't see the same issue. Can you zoom in and check the greyish part of the gripper is actually penetrates the cube? Or you can put a larger cube along size the arm and see if it can penetrate into the cube.

Thank you so much for your reply!

Yes, the greyish part of the gripper indeed penetrates the cube. Below are two examples showing different extent of penetration:

[image]

[image]

## larger cube or other USDs

I appreciate your suggestion to test with a larger cube. However, I realized this might not be an issue with the cube itself, because I have tried using cubes of different sizes and other objects (such as a beer bottle USD), while all of them end up being penetrated by the gripper.

[image]

## Stiffness and Damping

I suspect this problem could be related to either the Franka gripper’s stiffness/damping settings or an issue with Curobo.

Initially, I set the stiffness and damping to very high values: stiffness=80000 and damping=4000. With these settings, the gripper could pick up objects but would penetrate them. I also doubted if such high values align with real-world physical properties.

So later, I adjusted these parameters to match the settings from a franka panda example file. However, the resulting low values made it totally impossible to grasp objects, they would simply slip out from between the gripper fingers. Here is the relevant configuration code. So what stiffness should I set? A bigger one or a smaller one?

```python
robot = ArticulationCfg(
            prim_path=f"/World/envs/env_.*/Robots/{robot_name}",
            spawn=sim_utils.UsdFileCfg(
                usd_path=f"franka_panda_instanceable.usd", # I saved the official usd and reload it here
                activate_contact_sensors=False,
                rigid_props=sim_utils.RigidBodyPropertiesCfg(
                    disable_gravity=True,
                    max_depenetration_velocity=1,
                ),
                articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                    enabled_self_collisions=True,
                    solver_position_iteration_count=8,
                    solver_velocity_iteration_count=0,
                ),
                # collision_props=sim_utils.CollisionPropertiesCfg(
                #     collision_enabled=True, contact_offset=0.000, rest_offset=0.0
                # ),  
            ),
            init_state=ArticulationCfg.InitialStateCfg(
                pos=initial_position,  
                joint_pos={
                    "panda_joint1": 0.0,
                    "panda_joint2": -1.3,
                    "panda_joint3": 0.0,
                    "panda_joint4": -2.5,
                    "panda_joint5": 0.0,
                    "panda_joint6": 1.2,
                    "panda_joint7": 0,
                    "panda_finger_joint.*": 0.04,
                },
                joint_vel={".*": 0.0},  
            ),
            soft_joint_pos_limit_factor=1.0,
            actuators={
                "panda_shoulder": ImplicitActuatorCfg(
                    joint_names_expr=["panda_joint[1-4]"],
                    effort_limit_sim=87.0,
                    stiffness=80.0,
                    damping=4.0,
                ),
                "panda_forearm": ImplicitActuatorCfg(
                    joint_names_expr=["panda_joint[5-7]"],
                    effort_limit_sim=12.0,
                    stiffness=80.0,
                    damping=4.0,
                ),
                "panda_hand": ImplicitActuatorCfg(
                    joint_names_expr=["panda_finger_joint.*"],
                    effort_limit_sim=200.0,
                    stiffness=2e3,
                    damping=1e2,
                ),
            },
        )
```

##  doubts on Curobo 

Additionally, my grasping process is implemented using Curobo, though I have made some custom modifications to the environment, including using WorldConfig obtained from UsdHelper. I can't figure out where is the problem within curobo though.

## fixes on mesh doesn't solve problem

Finally, I acquired a USD file from an example that features collision colliders more tightly fitting the gripper’s visual geometry: [Grasp Editor-Getting Started-Download](https://docs.isaacsim.omniverse.nvidia.com/5.0.0/robot_simulation/grasp_editor.html) I then moved the fingers' mesh from this USD into my original USD file. While this modification still allowed my program to run, the fingers continued to penetrate objects:

[image]

Thank you again for your response!

RandomOakForest, 2025-10-23:
Following up, it seems this may have been resolved. I will close the issue for now, but feel free to reopen it if you need further help. Thank you.

WeisonWEileen, 2025-11-11:
> Following up, it seems this may have been resolved. I will close the issue for now, but feel free to reopen it if you need further help. Thank you.

what is your isaaclab version? I have same issue with isaaclab2.0.0&isaacsim5.0.0?

[image]

JyiDeng, 2025-11-14:
> > > > > Following up, it seems this may have been resolved. I will close the issue for now, but feel free to reopen it if you need further help. Thank you.
> > > > 
> > > > 
> > > > could you please tell me what is your isaaclab version? [@JyiDeng](https://github.com/JyiDeng) [@RandomOakForest](https://github.com/RandomOakForest) I have same issue with isaaclab2.3.0&isaacsim5.1.0?
> > > > <img alt="Image" width="830" height="307" src="https://private-user-images.githubusercontent.com/124496507/512487995-a4068e9f-523f-40de-a9bb-b3f062f20956.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjI4MzAzMTksIm5iZiI6MTc2MjgzMDAxOSwicGF0aCI6Ii8xMjQ0OTY1MDcvNTEyNDg3OTk1LWE0MDY4ZTlmLTUyM2YtNDBkZS1hOWJiLWIzZjA2MmYyMDk1Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTExMVQwMzAwMTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MDc2ODJkNjFjN2UyMGM0ODhhZWNkZTViODA0M2ViY2VlOGEzY2NlNTJiMGM3ZDFkMThkMmQ0ZDUyOWNkMTM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.1DEftL0EIz2ehn7sIruTnTgP8CtzdqmPvGxmLDmXPHE">
> > > 
> > > 
> > > I'm running Isaaclab 2.2.1 and Isaacsim 5.0.0.
> > 
> > 
> > Thanks for you reply! Did you solve your penetrate problem? If so, how did you solve your problem?
> 
> After I reduce the damping and stiffness, it will not penetrate. thanks!

Actually I increased dt in SimulationCfg, then increased stiffness and damping in franka robot's ImplicitActuatorCfg. I also tried to decrease rest_offset and contact_offset in franka's CollisionPropertiesCfg.

What I refered to are: 

- https://docs.isaacsim.omniverse.nvidia.com/4.2.0/simulation_fundamentals.html#contact-and-rest-offset
- https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/dev_guide/guides/articulation_stability_guide.html#reduce-simulation-timestep-for-complex-systems-and-closed-loops

Glad to see your problem solved!