isaac-sim/IsaacLab #3590: [Bug Report] tensor device error by --headless

taisi5446, 2025-09-29:
Hi Isaac Lab team,

I am encountering an `Incompatible device` error when running a standard reinforcement learning training script with Realsense cameras. The error is consistently thrown at every simulation step.

I am running the training with the following command:

./isaaclab.sh -p scripts/reinforcement_learning/skrl/train.py --task=Isaac-Twip-Rl-skrl-v0 --num_envs 1 --enable_cameras

This results in the following error at every step:

[Error] [omni.physx.tensors.plugin] Incompatible device of velocity tensor in function getVelocities: expected device 0, received device -1

**Debugging Findings:**
Through debugging, I've determined that the error is triggered during the execution of `super().step(render=render)` on **line 577** of `source/isaaclab/isaaclab/sim/simulation_context.py`.

**Additional Context:**
Notably, this error does not occur if I run the same command with the `--headless` flag. 

Could you please advise on how to resolve this issue? I can upload my working files (such as the task environment script) if necessary.

Thank you

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5.0
- Isaac Lab : 2.0.0
- OS: Ubuntu 22.04
- GPU: RTX 3070TI
- CUDA: 12.5
- GPU Driver: 555.42.06

PeterL-NV, 2025-09-30:
Hi, Can you try with the 2.2.0 release? The task you used is not available anymore so please try with a different task and report back if you still have issues.

PeterL-NV, 2025-10-16:
Please don't try this task, as it's not available in Isaac Lab 2.2.
From your output:
> |   54   | Isaac-Twip-Rl-skrl-v0                                         | isaaclab_tasks.direct.twip_rl.twip_rl_env:TwipRlEnv

But there's no such task under [`source/isaaclab_tasks/isaaclab_tasks/direct/`](https://github.com/isaac-sim/IsaacLab/tree/release/2.2.0/source/isaaclab_tasks/isaaclab_tasks/direct)

closed 2025-10-22 by taisi5446